ikarus23/MifareClassicTool #399: There is no.... Classic tag

Setrule, 2022-10-13:
Hello, i need to duplicate a Key for my portal but i just can't read my tag. I got an error message : error there is no mifare classic tags .
I tried many previous version of mifare classic tools and it got the same message on every version. 
My phone is a Poco f3 
Do i need to download somes files or anything before reading tags ?

ikarus23, 2022-10-22:
Hi. This app only works with MIFARE Classic tags. There are probably hundreds of different RFID tags. To me it sounds like your tag is just not a MIFARE Classic tag. Have you tried to detect the tag with other apps like NXP's TagInfo? Maybe it is even a 125KHz tag that can not be read by phones.

Setrule, 2022-10-22:
Yeah i tried to detect with nxp and it work. 
With mft i just can't read , he isnt even trying because he is asking me for classig tag as an error message before reading. I tried with various phones and it happened again.

ikarus23, 2022-10-22:
What did NXP's TagInfo say? What type of IC?

ikarus23, 2023-03-04:
Closing. Lack of feedback.

closed 2023-03-04 by ikarus23