ikarus23/MifareClassicTool #401: Sector trailer write bug

ArthurMao, 2022-10-20:
Hello, I've recently used this tool to clone a tag.
However, each time after cloning, the new tag differs slightly from the original tag.
The tool readout for the **Sector trailer** is "----------7F0788688888888888", but it cannot be written into the new tag.
Then I try changing "-" to 0150 (dash symbol HEX value) and writing into a new tag, which is completely identical to the original tag.
This appears to be a bug.

ikarus23, 2022-10-22:
I don't think there is a bug. If I understand you correctly you read the data `----------7F0788688888888888` as a sector trailer. This is incomplete data. `---` means this part is unknown. In a sector trailer, the first 6 bytes are key A. So you are missing key A. Incomplete data can not be written, because MIFARE Classic tags can only be written block by block. There is no such thing as half of a block. If you change the `---` to some hex value, you can write the dump. But remember, it has to be 6 bytes and it will be set as key A of the cloned tag.

ArthurMao, 2022-10-23:
Got it，I was misunderstood.

closed 2022-10-23 by ArthurMao