ikarus23/MifareClassicTool #402: Why are gen1 not supported ?

Youbze, 2022-10-29:
I apologize if this has already been asked (probably), but I can't find the answer anywhere ^^
You write in the readme "Some of them require a special command sequence to put them into the state where writing to the manufacturer block is possible. These tags are often called "gen1", "gen1a" or "UID"."
But why can't you implement sending these commands to write them ? Is it because of a limitation on the API on android ? If so you should maybe just add it to the readme so that it's more easily understandable :)

ArchangeGabriel, 2022-10-30:
> Is it because of a limitation on the API on android ?

Yes.

> If so you should maybe just add it to the readme so that it's more easily understandable :)

https://github.com/ikarus23/MifareClassicTool/pull/403 ;)

AkechiShiro, 2022-10-30:
@ArchangeGabriel is this limitation impossible to bypass ? I mean, if we're going for a custom `libnfc` and a custom ROM, is this impossible to support ? Or is it very hard to do ?

closed 2022-10-31 by ikarus23

ikarus23, 2022-10-31:
Thanks @ArchangeGabriel. I hope it is more clear now (especially the "why").

>is this limitation impossible to bypass ? I mean, if we're going for a custom libnfc and a custom ROM, is this impossible to support ? Or is it very hard to do?

Impossible, no, but I guess it might not be easy. Not sure what part has to be custom. Something in the Android NFC stack does not allow you to send an receive custom commands *without* checksum. Unfortunately the gen1 special commands do not use a checksum But Android always appends one and expects one.

A workaround for gen1 is to use MCT with an ACM122u reader connected via USB and with [External NFC](https://play.google.com/store/apps/details?id=eu.dedb.nfc.service) installed. This is possible because of the "Uses special activation for clone cards" feature of External NFC. It is not very stable, but I tested it and was able to make it work.
1) Activate the External NFC service and connect the reader
2) Do not leave the External NFC app
3) Present a tag to the reader, External NFC will most likely show you an app selection dialog
4) Choose MCT
5) Use the MCT (the instance spawned by External NFC), it should work now