ikarus23/MifareClassicTool #406: MCT stop reading with custom dictionary

schmiernippel, 2023-01-16:
I have made a custom dictionary with keys from all over the internet with7.5k keys.
When i use this, MCT stops reading on 50-90% without a error message.
I have tried to split the dic in 3 pieces of 2.5k keys, but its the same?
What is the problem?

ikarus23, 2023-03-04:
Hi. Interesting. Not sure what could cause that. On the other hand I'm not sure if the big key file is a good idea to start with. Doesn't take ages to go through all keys? And are you really sure that there is no error message? It might be already gone when you looked on the display again. (I only have shitty toast messages, sorry).

Anyways I think it could be some battery management system from Android. Or just a connection loss to the tag (you could try the auto reconnect function from the settings).

You could also consider buying a cheap RFID reader in order to crack the keys of the tag, instead of searching them using a key file.

ikarus23, 2023-08-17:
I will close this. I don't think there is much I can do here. Feel free to document any news regarding this topic, if you have some.