ikarus23/MifareClassicTool #407: Mifare Android no valid key found in key file

Tidoto, 2023-01-16:
So i recently bought nfc cards and tested one of them, it worked first try no problem, now i wanted to use another card, like the first time everything went well, until i tried to use it, it didnt work. i use Mifare for Android because i just cant get the PC version to work at all for me. After i wrote the nfc card and tested it when it didnt work, i tried the recover function of the program as it had worked with the previous card and that one is working perfectly fine, no matter how many times i tried, it can read some of it but not all of it, i knows what type is on the card but thats all. after a while of trying around myself i managed to get it to once again notice the card and it says its a new UID but when i try to read it, it says "no valid key found in key file" i checked the std.keys to find that absolutely nothing had changed, i cant read the card, i cant rewrite the dump onto the card and i also cant format the card, everytime it says "no valid key found in key file". ive been looking online to hopefully find help but i found nothing, ill provide the blank tag that i had made from when i could read the card and ill provide that dump that i tried to write onto it, please help me with this.
[UID_86D1E649_2023-01-16_18-48-30 blank tag.txt](https://github.com/ikarus23/MifareClassicTool/files/10428719/UID_86D1E649_2023-01-16_18-48-30.blank.tag.txt)
[dump i tried to write.txt](https://github.com/ikarus23/MifareClassicTool/files/10428722/dump.i.tried.to.write.txt)

ikarus23, 2023-08-17:
Hi. Sorry for the late response. Not sure if I didn't understand you correctly or if you miss some basic knowledge.

The blank tag you provided is indeed a blank tag. Nothing special there.

The dump you wrote contains data and keys. After writing the tag, you will no longer able to read/write all data of this tag. This includes the factory format. MIFARE Classic tags have keys. Think of them like passwords. You need the passwords to read/write data. The dump you wrote changed the keys. The `std.keys` file only contains the default factory keys of e.g. blank cards. In order to interact with the changed card, you need to create a key file will call the valid keys.

You can extract the keys form a dump and put them into a key file. There even is an automated way for that. Open the dump in the "Dump Editor". Then in the menu, press "Save Keys". You can now use the key file to read/write/reset this tag (as long as the permissions/access conditions allow that).

PS: Mifare Classic Tool Windows applications like https://github.com/xavave/MifareOneTool-English or https://apps.microsoft.com/store/detail/mifare-classic-tool/9NNZH1RH9RL0 have nothing to do with this project. They just use the same name and sometimes the same menu structure and/or pictures. They are not made by me.

Tidoto, 2023-09-30:
Oh hello!

I'm very sorry I forgot to reply.
On Android mifare doesn't have the permission that it needs to be able to
write everywhere, I have the same issue on my computer, or well had would
better. Up until now I didn't have the administrator account to my own
computer which was just unbelievable. I'm hoping that I will be able to
actually be able to access everything on these cards to write the tags
fully. I will test it on both my new phone and my computer in a bit, I hope
I'll be able to get it done today.

Once again I'm really sorry for not replying.

ikarus23, 2023-09-30:
Hi, don't worry! Take your time.

Not sure what the issue is with the computer, but your Android phone might be able to read the card as long as you have the correct keys for it. If you don't have the keys, you can crack them with tools like the Proxmark3.

Best of luck!

Tidoto, 2023-09-30:
This was quite a while ago so I need to get back to where I stopped trying.
I have an card reader and writer which is able to both read and write tags,
the problem is, I cannot write the entire card, from what I can remember
the problem was the first row of data in each sector, I'm not sure about
that description but there's a part of the cards I cannot write, this only
applies to the mobile version and not the PC version.
Going back to the keys stuff, that is no issue since the cards are being
written with specific keys, these keys are preset by the dumps that I write
onto the cards as the game uses specific keys to access the cards and it's
data, so I have a massive assembly of all the keys that can be written with
the dump.
The problem is really just that I can't write the entire tag for some
reason(on mobile).  If I remember correctly it was the UID? I'll have to
look into it more tomorrow.
Thank you for your patience!

ikarus23, 2023-10-02:
Oh, that would make sense. The first block of sector 0 ("manufacturer block", the very first line in a full dump) is read-only. On a normal card, you can not change this data. If you want to clone that data as well, you need a special card. They are called "magic cards"  or "CUID cards" (only "Gen2" version will work on Android). Some shops (like Aliexpress) advertise them as "Android compatible" or "MCT compatible".

Hope this helps.

Tidoto, 2023-10-02:
I had specifically bought cards that would allow me to write into the
manufacturer block as that is the most important data for the game.
I have also found a massive problem with my computer that would lead MIFARE
to give me false results and most of the time not even let me read tags,
turns out, the front two USB ports on my computer are completely useless
and can only transfer power.
with that problem quickly ignored and the usage of the back USB ports i can
now read the tags, on my phone i have not been able to read tags at all,
completely empty and unused tags dont work, used tags dont work either, i
will attempt to write the UID of the card ive managed to read(i have
written onto this card before im not sure if i was able to change the UID
but i know exactly what dump i attempted to write onto it so i will retry
that same dump).
i'll get back to you in a little bit
Thank you!

ikarus23, 2023-10-02:
Interesting. MIFARE Classic on Android should work fine, as log as you have a device with MIFARE Classic support. What device do you have? It should be even easier than Windows, because all driver/API things should be taken care of by Android.

Tidoto, 2023-10-02:
An arc122u connecter via USB to USBC adapter

ikarus23, 2023-10-03:
Sorry, I should have been more clear. I mean, what Android device are you using?

Also, what is your tool chain for Windows? libnfc with its tools (like `nfc-mfclassic`) should be perfect for the ACR122u. Personally, I never used it under Windows so far, but there is a README for that: https://github.com/nfc-tools/libnfc/blob/master/README-Windows.md Otherwise, you can set up a LInux VM. I think the Kali Linux distribution has all the tools already installed or at least ready to install (no compiling code needed).

Tidoto, 2023-10-03:
For this im using a Honor9X with android 10, i have written most of the
tags using it but for some reason it just doesnt work anymore, external NFC
says its connected to the reader even mentioning it by name but once i put
a tag on there(doesnt matter if blank or not) it gives me an error saying
theres no mifare classic tag.

oof that one is difficult, ive installed the version MWT v1.6.8424.42463
drivers and everything are also installed i just for some reason cannot
write any tags at all because it just instantly crashes once i tell it to
write.

Tidoto, 2023-10-03:
These are the tags I have:
https://www.ebay.de/itm/1-20-MCT-Gen-2-UID-Clone-Card-4-byte-NFC-Tag-Compatible-with-Mifare-Classic-1k-/154312693111
And I just can't write all blocks, only 45 out of 64 I don't know why, I
tried everything.
Blocks that failed to write:
3, 7, 11, 15, 19, 23, 27, 31, 35, 39, 33, 33, 41, 45, 49, 43, 47, 51, 55,
59, 63
im not sure which those are exactly but theyre not being written and
thats the problem.

ikarus23, 2023-10-03:
Yes, the cards look good.

Oh, the Honor9X does not have NFC and you are using External NFC to make it work with the ACR122U. I see! Yeah, that setup is unfortunately pretty unstable. The External NFC app seems out of active development for quite some time. If you or a friend of yours has an NFC enabled Android phone laying around, grab that. I will make things so much easier... 

>Blocks that failed to write:
>3, 7, 11, 15, 19, 23, 27, 31, 35, 39, 33, 33, 41, 45, 49, 43, 47, 51, 55, 59, 63

These are all sector trailers. Meaning the blocks where both keys and the permissions are stored in. Maybe the permissions of the card do not allow to change them? However this should work on a blank (new) card.

Tidoto, 2023-10-03:
its like that for every card, why cant i just nuke the data on it and call
it a day? i have all the keys that were set when i tried writing the card,
if i cant write those sectors i might aswell just snap the card since its
completely useless to me now.

ikarus23, 2023-10-03:
Like I said. To me it sounds like an issue with the software. The two things known to work pretty good:
1. NFC enabled Android phone with MCT (not using External NFC)
2. Compunter/Laptop with an external reader (like ACR122u) and libnfc as software
(I never tried MWT and have no idea how it works or if it works at all)

Tidoto, 2023-10-03:
I don't have any device with internal NFC.... the only one I had broke
because the battery started bloating up and when changing it... the screen
broke and the phone is unusable.

These are btw the results from mifare on windows, that's why I'm confused
as to why it's not working

Tidoto, 2023-10-11:
Hi,

You were saying that the mifare tool is installed on a Kali Linux
installation, where can I get one that for sure has everything that's
required?

Thanks in advance!

ikarus23, 2023-10-13:
Hi. Kali Linux has all the libnfc tools installed, including mfoc & mfcuk (I just checked).

If it is a modern MIFARE Classic card (e.g. EV1, with improved PRNG) you might need a special version of mfoc. You can find instructions here: https://github.com/nfc-tools/mfoc-hardnested

Also: try googling a bit. There are a lot of instructions out there about how to crack MIFARE Classic cards.

Tidoto, 2023-10-13:
yea.... youve said that before.
firstly, i dont have any clue where i can get a kali linux installation,
secondly, the only experience with linux ive had so far was setting up
raspberry pis,
ill for sure be able to figure out linux, but getting a linux installation,
especially a kali linux installation, that at the same time is actually
clean and not 99% maleware.
hence my question, as you seem to clearly have more knowledge on this than
me.
where can i get an installation thats actually safe to use?

ikarus23, 2023-10-15:
What do you mean? You can get the "clean" (original) kali at https://www.kali.org/get-kali/#kali-platforms
You can even use Virtualbox as VM system and get a preinstalled VM image for Virtualbox directly from kali.org (https://www.kali.org/get-kali/#kali-virtual-machines). This way you can run Linux inside a virtual machine on your Windows.

ikarus23, 2023-12-27:
I will close this for now (off-topic). Feel free to document your journey here, if you progress.

closed 2023-12-27 by ikarus23