ikarus23/MifareClassicTool #411: Ability to brute force and show key when found

sdushantha, 2023-02-20:
Thank you for this great application, I really enjoy using it!

I feature that I find to be missing is the ability to brute force the key of the card and show the key that is found.
Currently, MCT is able to attempt to crack the key by using a list but it doesn't show what the key was. Knowing what the key is would be useful

ikarus23, 2023-03-04:
Hi. Glad you like it. As the [General Information](https://github.com/ikarus23/MifareClassicTool#general-information) section of the readme states:
> There will be no "brute-force" attack capability in this application. It is way too slow due to the protocol.

Card key may have 12 hexadecimal digits (16 options per digit) 16^12 = 281474976710656 key combinations.
The app searchs 40 keys per second. This mean 281474976710656/40 = 7036874417766.4 seconds.
These are 117281240296.1067 minutes; which are 1954687338.268444 hours; so it would be 81445305.76118519 days.
So to be able to find a single key (out of typically 32 key), you might have to wait a maximum of **223137.8 years** to find it.

closed 2023-03-04 by ikarus23

ikarus23, 2023-03-04:
Oh, and also from the readme:
> Due to some weaknesses in MIFARE Classic, you can retrieve all the keys (A and B) of a tag with tools like the [Proxmark3](http://www.proxmark.org/) or normal RFID-Readers and some special software ([mfcuk](https://github.com/nfc-tools/mfcuk), [mfoc](https://github.com/nfc-tools/mfoc)).

ArchangeGabriel, 2023-07-19:
A bit late to the party, but the key are actually shown on the dump…