ikarus23/MifareClassicTool #412: Write dump file failed

btwang0512, 2023-02-23:
[test.zip](https://github.com/ikarus23/MifareClassicTool/files/10810145/test.zip)

btwang0512, 2023-02-23:
[image]

When the fitst digit of block data is  'C', the writing will be failed.

ikarus23, 2023-03-04:
Strange. Looks perfectly fine to me. Is it only on the `C` change? Are you able to write the dump at all? e.g. without changing anything at all?

btwang0512, 2023-03-04:
Thanks for your reply. Please see the attached file.Sent from my phone-------- Original message --------From: Gerhard Klostermeier ***@***.***>Date: Sat, Mar 4, 2023, 23:13To: ikarus23/MifareClassicTool ***@***.***>Cc: btwang0512 ***@***.***>, Author ***@***.***>Subject: Re: [ikarus23/MifareClassicTool] Write dump file failed (Issue #412)
Strange. Looks perfectly fine to me. Is it only on the C change? Are you able to write the dump at all? e.g. without changing anything at all?

btwang0512, 2023-03-04:
Sorry. The attached file is sent failed. I will resend it tomorrow.Sent from my phone-------- Original message --------From: ***@***.***: Sat, Mar 4, 2023, 23:59To: ikarus23/MifareClassicTool ***@***.***>, Gerhard
Klostermeier ***@***.***>, ikarus23/MifareClassicTool ***@***.***>Cc: Author ***@***.***>Subject: Re:
[ikarus23/MifareClassicTool] Write dump file failed (Issue #412)Thanks for your reply. Please see the attached file.Sent from my phone-------- Original message --------From: Gerhard Klostermeier ***@***.***>Date: Sat, Mar 4, 2023, 23:13To: ikarus23/MifareClassicTool ***@***.***>Cc: btwang0512 ***@***.***>, Author ***@***.***>Subject: Re: [ikarus23/MifareClassicTool] Write dump file failed (Issue #412)
Strange. Looks perfectly fine to me. Is it only on the C change? Are you able to write the dump at all? e.g. without changing anything at all?

btwang0512, 2023-03-07:
[write-dump.zip](https://github.com/ikarus23/MifareClassicTool/files/10907793/write-dump.zip)

Please see attached video file. If the first Digit of block is 'C', the dump write will be failed.

ikarus23, 2023-03-07:
There is no file. I get a 404 (not found) error.

btwang0512, 2023-03-08:
[image]
[image]
[image]
[image]

Sorry, the video file cannot be upload successfully. I cut some picture from video.

ikarus23, 2023-07-27:
Any news? Did you try the test version?

ikarus23, 2023-08-17:
Ok, I just pushed the ugly fix (https://github.com/ikarus23/MifareClassicTool/commit/ec3f0477ceb665b482ffb3194762dd76d3efcb8f) and hope it will help.

closed 2023-08-17 by ikarus23