ikarus23/MifareClassicTool #414: Add some features to diff tool

deunlee, 2023-03-25:
I've been using this app for years, and it would be great if a few more features were added to the diff tool.

- Show matching percentage of two selected dumps
- Option to hide same sectors

I can make a PR so let me know if it's okay.

ikarus23, 2023-08-17:
Hi! Sorry for the late response. Sounds great!

I would suggest just to insert a "Hide identical sectors"-checkbox below the dump chooser buttons (A) and to insert a simple text (e.g. "Difference between dumps: 0%") below the line (in the result area, B).
[image]

deunlee, 2023-08-22:
OK!
I'll prepare the PR.

ikarus23, 2023-09-16:
Merged. Thanks again!

closed 2023-09-16 by ikarus23