ikarus23/MifareClassicTool #437: some sectirs of gen2 tag had been broken after factory format.

alximiktik, 2023-10-05:
I have gen2 bracelet. Using MCT i can change UID. But there's a issues with factory format. After  factory format of gen2 tag, impossible to read some  sectors info. I have tried use proxmark3 , but couldn't fix it.  
[image]
Is it possible to fix it?

alximiktik, 2023-10-08:
I could find 3rd sector keys : A 000102030405 B 0A0B0C0D0E0F: 1,2 and 13 sectors key cant find let

ikarus23, 2023-10-16:
Strange.
> I could find 3rd sector keys : A 000102030405 B 0A0B0C0D0E0F: 1,2 and 13 sectors key cant find let

Do the found keys work? What is the content of the e.g. sector 3? Is it even writable? Maybe the factory format did not work because these sectors are read only?

ikarus23, 2023-12-28:
I will close this for now. Feel free to reopen this if you have more information and tried all the keys.

closed 2023-12-28 by ikarus23