ikarus23/MifareClassicTool #439: cannot read tag  in Huawei Harmony 4

jasperwt, 2023-10-11:
When reading tag, the software crashed. It always happened in recent days. The version is 4.0.5.

Little-King2022, 2023-10-12:
My software crashed too. How can I submit the log?

ikarus23, 2023-10-16:
Hi. Does this crash happen right away? Or after some time in the key mapping process? Did it work before (e.g. version 4.0.4)?

@Little-King2022: Do you also have a Harmony 4?

Little-King2022, 2023-10-20:
Yeap, my system is Harmony 4 too.
When I read the CPU card(In the china mainland its name is CPU card), the software can read the card id, but it cancrashed when I press the read button.

Little-King2022, 2023-10-20:
In the previous system version, this CPU card could be successfully read.

jasperwt, 2023-10-20:
> Hi. Does this crash happen right away? Or after some time in the key mapping process? Did it work before (e.g. version 4.0.4)?
> 
> @Little-King2022: Do you also have a Harmony 4?

Yes. I tried 4.0.3,4.0.4,4.0.5 and 4.0.10. They all failed. And I also found m1 cards worked while CPU card failed just like @Little-King2022 said.

jasperwt, 2023-10-20:
> Yeap, my system is Harmony 4 too. When I read the CPU card(In the china mainland its name is CPU card), the software can read the card id, but it crashed when I press the read tag button. In the previous system version 3.x, this CPU card could be successfully read. Additionally, standard m1 cards can be successfully read and write in both version 3 and 4. Only during the above process the software crashed down.

Hi, I tried version 3.2.0 but it also crashed. Would you mind telling me which version you used? Many thanks!

jasperwt, 2023-10-20:
> > > Yeap, my system is Harmony 4 too. When I read the CPU card(In the china mainland its name is CPU card), the software can read the card id, but it crashed when I press the read tag button. In the previous system version 3.x, this CPU card could be successfully read. Additionally, standard m1 cards can be successfully read and write in both version 3 and 4. Only during the above process the software crashed down.
> > 
> > 
> > Hi, I tried version 3.2.0 but it also crashed. Would you mind telling me which version you used? Many thanks!
> 
> 我记不清了，今年上半年的Harmony 3.x版本是可以正常读取cpu卡的。上个月升级到4.0之后就无法读取cpu卡。 I can't remember clearly. The Harmony 3. x version in the first half of this year was able to read CPU cards successfully. However it is unable to read CPU card after upgrading to 4.0 last month.

The same situation as you mentioned. I mistakenly thought you were referring to SOFTWARE version 3. x

Little-King2022, 2023-10-20:
I'm sorry for causing a misunderstanding between the HarmonyOS version and MifareClassicTool version. However, if possible, you can try downgrade your phone to the HarmonyOS 3.x, in which I successfully read my CPU Card.

ikarus23, 2023-10-25:
Oh, in that case the issue is related to the OS version. In that case I don't think there is much I can do. Sometimes vendors change their NFC driver/firmware with an OS update. This happened to some Samsung devices as well.

Unless I have access to a device with HarmonyOS (which I don't), I can not reproduce this issue. Maybe someone with an HarmonyOS device and some Android debugging skills can look into this.

Chinaholmesfans, 2023-10-26:
When reading tag, the software crashed too on my phone.  [Huawei Harmony 4]

ikarus23, 2023-10-29:
Does [this testing version}(https://tests.icaria.de/MifareClassicTool-4.1.0-test1.apk) work for you?

ikarus23, 2024-01-02:
I made a newer testing version. Could you please try it: https://tests.icaria.de/MifareClassicTool-4.1.0-test7.apk

jasperwt, 2024-01-03:
> I made a newer testing version. Could you please try it: https://tests.icaria.de/MifareClassicTool-4.1.0-test7.apk

Sorry, I missed your testing version in Oct 30. I tested the test7.apk this day and it works. Very happy to see this fantastic software work again in my phone. Many thanks to you!

ikarus23, 2024-01-03:
Great! Thank you for testing.

closed 2024-01-03 by ikarus23