ikarus23/MifareClassicTool #441: Google Pixel 8 Pro Problem

monelli, 2023-10-17:
Phone: Google Pixel 8 Pro
App version 4.1.0

Display Tag Info works well.
Start Mapping and Read Tag doesn't work.

Output:
New tag found (UID:XXXXX)
7 keys loaded. Creating key map This might take a while
No valid key found in chosen key file

Same Tag and same key file in another phone work.

Thank you!

ikarus23, 2023-10-17:
Hi! Thanks for reporting. I have no idea what could cause this. Unfortunately, I do not have a Pixel 8 Pro laying around to recreate the issue.

I will try to reach out to some people who may have a Pixel 8 (Pro). Of course, if you are familiar with Android debugging you can try it your self.

ikarus23, 2023-10-17:
Anything special with this tag? Does it have default keys? Did you try other apps (e.g. NFC TagInfo by NXP, full scan)?

monelli, 2023-10-17:
I don't know android debug, but I'll try to experiment.
I tried the other recommended application, but the behavior is strange.
See image:
[https://i.imgur.com/i925AgQ.png](https://i.imgur.com/i925AgQ.png)
first line pixel 6
second line pixel 8

I inserted all the keys.
I obtained the keys with the milazy, mfoc, mfcuk tools.
I confirm that the keys work because with the pixel 6 I read and write correctly.

mansellrace, 2023-10-20:
Hi! I also have a Pixel 8 Pro and I confirm that I cannot interact with a tag that I have always used on any other phone. The association with the keys is not recognized

paloindici, 2023-10-23:
Hi !
I have the exact same problem on Pixel 8 Pro. Try on several badges, none work...

ikarus23, 2023-10-23:
Does any of you have a factory formatted (empty) MIFARE Classic tag? I would be interested to know if an app like NFC TagInfo by NXP can read it (using the "full scan" feature/tab).

crossproduct, 2023-10-24:
Same here, does not work on pixel 8 pro. Also had this working perfectly fine on Pixel 6 pro.

paloindici, 2023-10-24:
> Does any of you have a factory formatted (empty) MIFARE Classic tag? I would be interested to know if an app like NFC TagInfo by NXP can read it (using the "full scan" feature/tab).

Hi,
a blank badge seems to be read from the application you mentioned.

[image]

paloindici, 2023-10-24:
> Does any of you have a factory formatted (empty) MIFARE Classic tag? I would be interested to know if an app like NFC TagInfo by NXP can read it (using the "full scan" feature/tab).

But this same badge also seems to be read from MCT

[image]

ikarus23, 2023-10-25:
Damn. I thought we might be one step closer. But if it can be read by either application, I don't know for sure that the bug is in MCT. It could be somewhere in Android or drivers of a Google Pixel 8 Pro.

ikarus23, 2023-10-29:
Does [this testing version](https://tests.icaria.de/MifareClassicTool-4.1.0-test1.apk) make any difference?

mansellrace, 2023-10-29:
> Does [this testing version](https://tests.icaria.de/MifareClassicTool-4.1.0-test1.apk) make any difference?

For me, nothing new, the problem remains

ikarus23, 2023-10-29:
Thanks for testing.

Other idea: Could you try the "Retry authentication" option and set it so something like 5?

mansellrace, 2023-10-30:
> Thanks for testing.
> 
> Other idea: Could you try the "Retry authentication" option and set it so something like 5?

Unfortunately it doesn't work anyway

imfloflo, 2023-11-03:
Fyi same thing with pixel 8. 
Pixel 5 read the tag and find the keys. 
Try to import Key files and read it from pixel8 nothing work.

imfloflo, 2023-11-08:
Just update my Pixel 8 to latest OTA from november 2023 : shiba-ota-ud1a.231105.004-49029a3a.zip 
MCT no works great with tag 😃

changelog [HERE](https://support.google.com/pixelphone/thread/242705137/google-pixel-update-november-2023?hl=en)

NFC
Fix for issue occasionally causing NFC and related services to have instability in certain conditions

ikarus23, 2023-11-11:
Thanks @imfloflo! This is great news. So it was a ROM/driver issue after all (which makes a ton of sense). I hope this update will roll out fast.

I will close this issue for now. Feel free to to open another issue if there issues with the app itself.

monelli, 2023-11-13:
I updated my Pixel 8 Pro to the same version (UD1A.231105.004) but I still have the same problem...

paloindici, 2023-11-13:
I feel like the NFC chip is quite sensitive. Some badges are read, others are not. And those that are read don't work all the time. The only one I need to read every day I can't read it. 
I also did update UD1A.231105.004, before this update I could only read blank badges.
It's a shame for this type of phone....

CleverManDev, 2023-12-07:
The problem not gone with the latest updates for Pixel 8 Pro.
I'm trying read tag but get message "no valid key found in choosen key file". The problem was on the november patch and now this problen on the latest december patch.

ikarus23, 2023-12-08:
@CleverManDev But what are the correct keys? Do you know them? Do you know that they are in the file(s)?

CleverManDev, 2023-12-11:
> @CleverManDev But what are the correct keys? Do you know them? Do you know that they are in the file(s)?

Google Pixel 7 Pro reads this tags without problems.
The result of reading attached  to this post (in zip).
[UID_34B22531D78436_2023-12-11_12-39-56.zip](https://github.com/ikarus23/MifareClassicTool/files/13633577/UID_34B22531D78436_2023-12-11_12-39-56.zip)

CleverManDev, 2023-12-28:
Any ideas with this problem?

alexg-k, 2024-01-25:
I have the same symptoms with a Pixel 8 (non-pro). The phone has the latest update from January..
This app and one I developed myself to read Mifare Classic cards worked on my Pixel 5, but stopped working on the Pixel 8 using the same cards.

ikarus23, 2024-01-26:
Hi. Sorry, still not idea what can cause this. I really need to get my hands on a Google Pixel 8...

CleverManDev, 2024-01-26:
> Hi. Sorry, still not idea what can cause this. I really need to get my hands on a Google Pixel 8...

Maybe we can help you with some log files? I have root privileges on my Pixel 8 Pro if it can help.

ikarus23, 2024-01-27:
Hi @CleverManDev. As a first try you could open the dump in the editor, then press on the menu an select "Save Keys". After this, go back to reading the tag and only select the new key file.

alexg-k, 2024-02-01:
I believe it could be a problem on Android-level, since other apps like nfcTags and my custom app are also not able to read a Mifare card on my Pixel 8. They worked before on my Pixel 5.

CleverManDev, 2024-02-06:
> Hi @CleverManDev. As a first try you could open the dump in the editor, then press on the menu an select "Save Keys". After this, go back to reading the tag and only select the new key file.

Opened the dump from Pixel 7 Pro, saved keys and readed std.keys and extended-std.keys.
[image]
[UID_34B22531D78436_2024-02-06_15-06-59.zip](https://github.com/ikarus23/MifareClassicTool/files/14179502/UID_34B22531D78436_2024-02-06_15-06-59.zip)

ikarus23, 2024-02-13:
I've just bought an Pixel 8. I can confirm. It can not read a genuine MIFARE Classic EV 1 card (with valid originality signature by NXP). It does not work on MCT, NXP TagInfo and NFC tools.

A quick debug shows: the error seems to be withing Android. Using the [MifareClassic.authenticateSectorWithKeyA(...)](https://developer.android.com/reference/android/nfc/tech/MifareClassic#authenticateSectorWithKeyA(int,%20byte[])) function from the Android API returns `false` even if the key is correct.

ikarus23, 2024-02-13:
I've tried to contact the Pixel 8 customer service, because someone said the general Android bug tracker is not for device specific issues. Let's see how this goes.

Not sure why it works for @imfloflo. Sure, there might be different Android versions depending on regions. But e.g. different NFC drivers for the same device... odd. Not sure if `shiba-ota-ud1a.231105.004-49029a3a.zip`contained something special which not all users got.

paloindici, 2024-02-13:
I also posted on the pixel forum just now to see if anyone has an idea or if Google sees the message...

https://support.google.com/pixelphone/thread/258613955/mifare-cards-are-not-read-on-pixel-8-and-pixel-8-pro

ikarus23, 2024-02-14:
@paloindici
>https://support.google.com/pixelphone/thread/258613955/mifare-cards-are-not-read-on-pixel-8-and-pixel-8-pro

I get a "This page is no longer available" for this link.

@alexg-k: Tanks for the feedback. This makes it look more like a NFC driver/firmware issue

I've also installed the `231105` version of stock Android, which was reported to work, but with no luck. Even with this version I was not able o read MIFARE Classic tags.

paloindici, 2024-02-14:
@ikarus23 
They declared my post Off Topic (I had put it in the communication section) I modified it in another I will see if they like it better...

[image]

imfloflo, 2024-02-14:
@ikarus23 I just try again with my P8 on latest version to read a NFC Aztec badge.
Try to read with the extended-std.keys (1766keys) as my previously save keys file was not reading it, but it was not working. 
It seems something is broken again. :/

ikarus23, 2024-02-15:
Good news: I'm now able to read MIFARE Classic tags with my Pixel 8!
Bad news: I don't know why.

Let's try to recreate it. Here is what I did:
1. Enable developer mode: https://developer.android.com/studio/debug/dev-options#enable
2. Enable (all?) NFC developer debugging options ("NFC stack debug log", "NFC verbose vendor debug log", "NFC NCI  unfiltered log")
3. Restart (some of the options above require this)
4. Try to read MIFARE Classic tag. For me it worked!
5. Disable all options
6. Still works!

Some things that would be interesting:
1. Does it work for you too?
2. Does it also work if only 1 of the 3 NFC debug options is used?
3. Which option is the one that make it work
4. And of course: Why?! Why is it fixed after that? What changed?

ikarus23, 2024-02-15:
By now I've locked down my bootloader again, reverted everything to stock, did a full factory reset, aaaaaand it still works! I guess, yay?! ¯\_(ツ)_/¯

paloindici, 2024-02-15:
Great progress !
I don't have anything to try now, I'll try this tomorrow 👌
Thank you for this research 👍

paloindici, 2024-02-16:
I did all the operation step by step with a restart and a reading test between each step and it still does not work for me once everything is activated. on the other hand, have you tried with a badge that has default keys? because if all the blocks have a default key it works. but does not work if at least one of the keys is not in the default key.

ikarus23, 2024-02-16:
Uff, yes, I only tested against a card with default keys. I tired different dumps and tags and my results got even worse.

* I have trouble using a original MIFARE Classic EV1 tag. Sometimes it works sometimes it don't. I only got it to work on empty tags with default keys
* I have way better results using a Gen1a,  CUID Gen2 and Gen3 magic tag. Even when having different keys, things work.
* Even using a (old) original MIFARE Classic tag (not EV1) thinks work with different keys.

Are there only issues with MIFARE Classic EV1 tags?

alexg-k, 2024-02-17:
I can confirm that the symptoms persist after the developer options are enabled on my custom rom.

monelli, 2024-02-18:
Hi, I confirm that even by activating NFC debugging I cannot read the tag.
I confirm that I have the problem on MIFARE Classic EV1 tag. (I haven't tested other tags).

Holbormon, 2024-02-23:
Hi, just wanted to add my 2 cents to the debugging effort:

Example of a tag that is read just fine:
[image]

Example of a tag that fails to be read (w/ correct keys, tested against same tag on another phone):
[image]

Note that [Metrodroid](https://github.com/metrodroid/metrodroid) also seems to be affected by this inability to read.

ikarus23, 2024-02-23:
@Holbormon Thanks. Unfortunately MCT's tag detection is not the best. NXP's TagInfo is way better (https://play.google.com/store/apps/details?id=com.nxp.taginfolite). It might be giving you information about the chip used in the tag. So far, MIFARE Classic EV1 chips seem to be the main issue. (But their could be more.)

Holbormon, 2024-02-28:
@ikarus23 Using that other app I can confirm that the affected tag is indeed a Mifare Classic MF1S50 (a.k.a EV1). Aside from mentioning that my chinese Mifare Classic 1K keyfobs are working just fine, there's not much else I can say. Hope we crack this one soon :muscle:

alecor450, 2024-05-14:
@ikarus23 here are more details on the issue:
Hardware used: Pixel 8 (Android 14), MIFARE Classic EV1 (MF1S70).
The tag is successfully read with a Samsung Galaxy S20 (Android 11), so it's not a defective tag.

The call to `authenticateSectorWithKeyA()` in `MCReader` ends up calling `mTag.getTagService().transceive()` in `BasicTagTechnology` [1].
From there, the Android Studio debugger cannot go further.
The result calls `TransceiveResult.getResponseOrThrow()` and returns `default: throw new IOException("Transceive failed");` [2].

I enabled **NFC stack debug log** and **NFC NCI unfiltered log** in the developer options. It gives the following log lines when the tag is discovered:
```
NativeNfcTag         com.android.nfc  D  Connect to a tech with a different handle
libnfc_nci           com.android.nfc  E  [ERROR:rw_mfc.cc(530)] RW_MfcDetectNDef
libnfc_nci           com.android.nfc  E  [ERROR:rw_mfc.cc(891)] rw_mfc_readBlock: RW_MFC_SUBSTATE_WAIT_ACK
NativeNfcTag         com.android.nfc  D  Check NDEF Failed - status = 3
libnfc_nci           com.android.nfc  E  [ERROR:rw_mfc.cc(530)] RW_MfcDetectNDef
libnfc_nci           com.android.nfc  E  [ERROR:rw_mfc.cc(891)] rw_mfc_readBlock: RW_MFC_SUBSTATE_WAIT_ACK
NativeNfcTag         com.android.nfc  D  Check NDEF Failed - status = 3
NativeNfcTag         com.android.nfc  D  Starting background presence check
ActivityTaskManager  system_server    W  startActivity called from non-Activity context; forcing Intent.FLAG_ACTIVITY_NEW_TASK for: Intent { act=android.nfc.action.TECH_DISCOVERED flg=0x20000000 cmp=de.syss.MifareClassicTool/.Activities.KeyMapCreator (has extras) }
ActivityTaskManager  system_server    I  START u0 {act=android.nfc.action.TECH_DISCOVERED flg=0x20000000 cmp=de.syss.MifareClassicTool/.Activities.KeyMapCreator (has extras)} with LAUNCH_MULTIPLE from uid 10338 (realCallingUid=1027) (BAL_ALLOW_VISIBLE_WINDOW) result code=3
```
Which corresponds to the file `rw_mfc.cc` [3].

Note that the NXP TagInfo App [4] also fail to read the tag.

Also, the following error log is triggered for every call to `authenticateSectorWithKeyA()`:
```
[ERROR:NfcTag.cpp(1533)] NfcTag::getTransceiveTimeout: invalid tech=0
```
Happening in `NfcTag.cpp` [5]. The expected `tech` should be 8 if I am not mistaken [6].

My conclusion is: unless there is a way to fix it by providing custom parameters to the NFC interface, it has to be fixed in the Android codebase.

[1] https://android.googlesource.com/platform/frameworks/base/+/refs/heads/android14-release/core/java/android/nfc/tech/BasicTagTechnology.java#149
[2] https://android.googlesource.com/platform/frameworks/base/+/refs/heads/android14-release/core/java/android/nfc/TransceiveResult.java#52
[3] https://android.googlesource.com/platform//system/nfc/+/refs/heads/android14-release/src/nfc/tags/rw_mfc.cc#530
[4] https://play.google.com/store/apps/details?id=com.nxp.taginfolite&hl=en_US
[5] https://android.googlesource.com/platform/packages/apps/Nfc/+/refs/heads/android14-release/nci/jni/NfcTag.cpp#1533
[6] https://android.googlesource.com/platform/frameworks/base/+/refs/heads/android14-release/core/java/android/nfc/tech/TagTechnology.java#140

ikarus23, 2024-05-16:
Wow, thanks for the research! I tried to reach out the Pixel support. So far, they were not very helpful. I referenced this issue multiple times and did send them the logs files. Communication came to a halt and after the last update the just asked: "is it fixed now?". Not sure how to continue from here...

nicof38, 2024-06-28:
Tested on android 15 beta2, and still the same faulty behavior on pxiel 8 pro :(

nicof38, 2024-07-19:
tested with today release of the newest android 15 beta, and now it works : AP31.240617.009
@ikarus23 can you confirm with your pixel 8 ?

archi, 2024-10-07:
Uh, careful: I tried this with a card I rarely use, and now that card fails to read:

1. Recovered keys using proxmark3 (using the improved attack types)
2. Manually copied all the keys into MFC
3. Tried to read the card with my Pixel 8 pro, no success
4. pm3 now fails

Error in pm3:

```
[usb] pm3 --> hf search                       
 🕒  Searching for ISO14443-A tag...[#] BCC0 incorrect, got 0x00, expected 0x32
[#] Aborting
[!] ⚠  No known/supported 13.56 MHz tags found
```

Previously, the card id'd as:

```
[usb] pm3 --> hf search
 🕗  Searching for ISO14443-A tag...          
[=] ---------- ISO14443-A Information ----------
[+]  UID: [redacted]   ( ONUID, re-used )
[+] ATQA: 00 04
[+]  SAK: 08 [2]
[+] Possible types:
[+]    MIFARE Classic 1K
[=] proprietary non iso14443-4 card found, RATS not supported
[=] 
[+] Prng detection....... hard
[=] 
[=] --- Tag Signature
[=]  IC signature public key name: NXP MIFARE Classic MFC1C14_x
[=] IC signature public key value: [redacted]
[=]     Elliptic curve parameters: NID_secp128r1
[=]              TAG IC Signature: [redacted]
[+]        Signature verification: successful
```

ikarus23, 2024-10-08:
HI @archi. I guess your issue is off topic from this issue, but it looked like you have an issue with the BCC (`BCC0 incorrect, got 0x00, expected 0x32`). The BCC on a 4 byte UID tag is the typically the 5th byte in block 0. This app has an BCC calculator under tools. Maybe this helps you debugging your issue. Anyway, this is not the right place to discuss it. Feel free ask in the RFID Hacking Discord server by iceman: https://discord.com/invite/iceman I also hang out there from time to time :)

ikarus23, 2024-10-15:
Thanks for the heads up. I checked an saw that Android 15 is also available now on my Pixel 8. Just installed it and did some basic tests. All seems to be working now! Finally! All this mails to support and waiting has payed of.

dpeddi, 2024-11-14:
Maybe pixel 8 should be added to compatibile devices?