ikarus23/MifareClassicTool #446: Import custom dictionary file size is limited

DevilHunterXX, 2023-11-15:
After testing, the upper limit of the size of a single dictionary file is 50kb

I hope the dictionary size can support a larger size

Thanks

ikarus23, 2023-11-19:
Interesting. I don't remember putting a limit there. But could be. Was there an error message? I mean, at a certain file size it might be the wrong approach anyways. Cracking the keys with an USB NFC reader or a Proxmark is way more efficient. I will have a look regardless.

DevilHunterXX, 2023-11-20:
> Interesting. I don't remember putting a limit there. But could be. Was there an error message? I mean, at a certain file size it might be the wrong approach anyways. Cracking the keys with an USB NFC reader or a Proxmark is way more efficient. I will have a look regardless.

Error: Some key(s) contain invalid data (not hex)

In fact, I have PN532, ACR122U, and Proxmark3  , and they are indeed faster. However, a single phone is easier to carry around or use for emergencies.

ikarus23, 2023-12-08:
The error message does not make sens with regard to the issue. Might be a bug or some memory limitations.
Or the error message is right. Are you willing to share the key file? Maybe there is some wrongly formatted data there.

ikarus23, 2023-12-27:
Any news? Are you willing to share the file?

ikarus23, 2023-12-27:
Just checked the code. I can not find any upper limit. So I think the error message is right. Your file contains something, that is not correctly formatted.

ikarus23, 2023-12-27:
Feel free to reopen this issue if you can share the key file. Otherwise it will be hard to tell what the issue is.

closed 2023-12-27 by ikarus23

DevilHunterXX, 2023-12-27:
[Uploading key.txt…]()

DevilHunterXX, 2023-12-27:
[key.txt](https://github.com/ikarus23/MifareClassicTool/files/13780488/key.txt)

ikarus23, 2023-12-28:
Well, look at the very end of the file! This is not a correct format. The error message was right!
[image]

DevilHunterXX, 2023-12-29:
[key_split00.txt](https://github.com/ikarus23/MifareClassicTool/files/13791371/key_split00.txt)
what about this key file ?
[image]

ikarus23, 2023-12-29:
Maybe you have to do some reading on Mifare Classic. A key for a Mifare Classic tag is 6 bytes long. The key is represented as hexadecimal (0-9, a-f). The key file of MCT should only contain keys formatted in that way.

Looking at the file there are two lines which contain invalid data. There is just `*` symbol.
[image]