ikarus23/MifareClassicTool #451: Android 14 support

jalenzz, 2023-12-07:
It seems not support Android 14

I uesd version 4.1.0，the read tag and write tag did not work

ikarus23, 2023-12-08:
Hi. There are users with Android 14 and no issues. So let's dig deeper. What device are you using? Have you tried other apps like NXP's TagInfo?

jalenzz, 2023-12-09:
I have tried NXP's TagInfo, it can successfully read the data of card.

I user MEIZU 21, Flyme 10.5, Android 14. I record a video

https://github.com/ikarus23/MifareClassicTool/assets/57910033/4cd3bc59-b91b-443f-9c3b-8c8621a98755

jalenzz, 2023-12-09:
read tag and write tag have this problem

but other functions like clone UID works well

ikarus23, 2023-12-27:
Hi. Can you try [this testing version](https://tests.icaria.de/MifareClassicTool-4.1.0-test2.apk)? There is a fix for Android 14 in there.

jalenzz, 2023-12-27:
It works now!
Thanks!

closed 2023-12-27 by jalenzz