ikarus23/MifareClassicTool #453: Compatibilty w Samsung Galaxy Tab Active 2

pwarner13, 2023-12-22:
Hi ÷ long time user, first time  posting.  Apologies in advance if this is the wrong place for this...

I have this tablet and I have successfully copied cards at my apartment building using Chinese magic cards that specifically stated android compatibility in the sales description.

For reasons unknown, this device is listed on your incompatible devices list.  It is also listed on the compatible devices matrix linked at the bottom of the afore-mentioned incompatible devices page.  http://www.shopnfc.it/en/content/7-nfc-device-compatibility  with all three columns checked.  I paid/donated a little over a year ago.  When I try to download on google play it says "incompatible w your device" but the link above and my personal experience says otherwise.  Should it be removed from the incompatible list, or am I overlooking something?

Thank you!

ikarus23, 2023-12-26:
Hi!

Thanks for reporting this. If you can confirm that the Samsung Galaxy Tab Active 2 is working for reading and writing cards, then I will update things. The device being on the "incompatible list" has most likely a historic reason. When first tested it was probably not working. Things got fixed later down the line and now it is working. I see this for a lot of devices. Even the other way around (stop working after Android (driver) update).

Have a nice day!

ikarus23, 2023-12-27:
I added the Samsung Galaxy Tab Active2 to the compatible devices and removed from the block list of Google Play. You should be able to install it from there now.

closed 2023-12-27 by ikarus23