ikarus23/MifareClassicTool #465: Don't start on Saumsung S24 Ultra (Android 14)

akirax2304, 2024-01-25:
The App don't start, it shows Welcome screen and when you push "Accept" it closes.

ikarus23, 2024-01-26:
Hi. Can you specify the dialog?
1. "This application is dedicated to users who have at least basic familiarity with the MIFARE Classic technology."
2. "MIFARE Classic Tool has been updated!"
3. "NFC is not enabled"

akirax2304, 2024-01-26:
Please check this video to see the problem:
https://youtu.be/AVJiCZv68qQ?si=3VGttOOmpsS5oTrY

Sorry for my English.

edusperoni, 2024-01-27:
I'm having the exact same issue. Here's the logcat error:
```
E  FATAL EXCEPTION: main
                                                                                                    Process: de.syss.MifareClassicTool, PID: 14259
                                                                                                    java.lang.NullPointerException: Attempt to get length of null array
                                                                                                    	at de.syss.MifareClassicTool.Common.hasMifareClassicSupport(Common.java:852)
                                                                                                    	at de.syss.MifareClassicTool.Activities.MainMenu.runStartUpNode(MainMenu.java:196)
                                                                                                    	at de.syss.MifareClassicTool.Activities.MainMenu.runStartUpNode(MainMenu.java:192)
                                                                                                    	at de.syss.MifareClassicTool.Activities.MainMenu.lambda$createFirstUseDialog$1$de-syss-MifareClassicTool-Activities-MainMenu(MainMenu.java:383)
                                                                                                    	at de.syss.MifareClassicTool.Activities.MainMenu$$ExternalSyntheticLambda7.onCancel(Unknown Source:2)
                                                                                                    	at android.app.Dialog$ListenersHandler.handleMessage(Dialog.java:1905)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:106)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:230)
                                                                                                    	at android.os.Looper.loop(Looper.java:319)
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:8918)
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method)
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:608)
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1103)
```

ikarus23, 2024-01-27:
Thanks for debugging and fixing this issue @edusperoni. I was not aware, that the folder does no longer exist on some phones.

The fix will be in the next release!

closed 2024-01-27 by ikarus23

edusperoni, 2024-01-27:
I'm also quite surprised. I couldn't find anything regarding the removal of 32 bit libraries on any Android dev resource. I'm wondering if this is a Samsung thing or if other manufacturers will also do it.

PR #466 (2024-01-27): feat: add S24 support and handle missing lib folder
Fixes #465