ikarus23/MifareClassicTool #469: External NFC and a phone that is not capable of NFC

HeXpp, 2024-02-14:
I have a Samsung Galaxy S3 that is not capable of NFC (I don't know why). I rooted it and installed external NFC, but iI can't bypass the pop-up that doesn't let me enable read/write functions. Could anyone help me?

Best regards,
HeXp

ikarus23, 2024-02-15:
Hi. Do you have more information about your Galaxy S3? Long time ago people reported that this NFC enabled device is capable of reading/writing MIFARE Classic tags (https://github.com/ikarus23/MifareClassicTool/blob/master/COMPATIBLE_DEVICES.md)

HeXpp, 2024-02-15:
> Hi. Do you have more information about your Galaxy S3? Long time ago people reported that this NFC enabled device is capable of reading/writing MIFARE Classic tags (https://github.com/ikarus23/MifareClassicTool/blob/master/COMPATIBLE_DEVICES.md)

Hi, thanks for replying. My S3 hasn't got a battery so it's connected directly to a wire. I think the NFC chip is in the battery itself since it has 4 pins, so no battery, no NFC. External NFC works ok with my ACR122u connected through OTG, so there's any way to bypass the screen that says "NFC is not enabled", so I can use my ACR122U?

I don't want to use the internal NFC chip, just the ACR122U.

P.S: Sorry for grammar errors, i'm not a native English speaker.

Best regards,
HeXp

ikarus23, 2024-02-15:
Hmm, I don't think the NFC chip is in the battery, but most of the time the the NFC antenna is in the back plate of the phone cover. Regardless, I understand that you want to use External NFC.

Maybe one stupid question: Do you have NFC enabled? I mean in the normal system dialog of Android. I think if your device has NFC you need to enable NFC (even if there is no antenna connected in your case). As long as it's on, you should be able to just use your external NFC reader.

But in general: Using External NFC with MCT is buggy at times. It has been a long time since I last tested things. Sorry, there is no big use base for this use case.

ikarus23, 2024-04-03:
Closed. Lack of feedback.

closed 2024-04-03 by ikarus23