ikarus23/MifareClassicTool #471: Succesful clone message but destination card is not actually written

sesardelaisla, 2024-02-22:
Hello,

I have purchased the following wristbands for cloning purposes:

https://www.amazon.com/gp/product/B098X38T98/

They are supposed to be Sector0 Block0 writable.

I can read both the source tag and the wristband. Cloning process seems to work fine because the message "Data successfully written" is displayed. However, if I re-read the wristband, content is not the same than source tag (UID does not change). All other

You can see below the tags info for reference:

**SOURCE**
[image]

**DESTINATION (WRISTBAND)**
[image]

I am doing this with a Samsung J5 (SM-J530F), MifareClassicTool Version 4.2.2.

Is this smartphone model really able to write? Why is the successfully message displayed if the destination tag is not the same? Doesn't such process compare to source reading after cloning to check whether the process was performed fine?

Any advise will be appreciated. Happy to provide further info if necessary. Thanks!

ikarus23, 2024-02-22:
Somebody in the comments says
> Used a Proxmark to clone my convention hotel key onto a wrist-band so I didn't have to worry about losing the keycard when in costume. Worked perfectly following the directions for a Gen1 magic tag!

This looks bad. Gen1(a) tags are not compatible with Android. You can only change the UID with dedicated hardware (USB NFC reader, Proxmark3, FlipperZero, etc.)

sesardelaisla, 2024-02-22:
> Do you have another Android device to compare?

I am afraid I don't.

> Did you use the "Clone UID" tool? Or did you write the whole tag?

The whole tag. Will try to clone the UID only and will give you feedback. Thanks!

> Do you have a Proxmark3?

Nop. I've ordered a few USB programmers from Amazon which will arrive tomorrow. I hope any of them work! In addition, I've also ordered [this device from AliExpress](https://es.aliexpress.com/item/1005005872978478.html) which will be delivered in a week or so. I hope any of these options work.

I will update this thread once I have the chance to do further tests for anyone's reference.

Thanks for your feedback, @ikarus23. Cheers!

sesardelaisla, 2024-02-22:
> The whole tag. Will try to clone the UID only and will give you feedback. Thanks!

Couldn't wait and I just tried the clone UID option. ;) It seemed to work. However, I read the wristband tag after programming and it returns the original UID.

As mentioned in my previous post, will try with the devices I ordered and will give further feedback.

sesardelaisla, 2024-02-23:
Finally, I could clone the tag with this device:

https://www.amazon.es/dp/B0C5X2RVR1

And this software:

http://www.nsccn.com/web/soft/down_zhcn.html

Process is very straight forward and it works like a charm.

I guess the Samsung J5 hardware cannot write for whatever reason. Cheers!

closed 2024-02-23 by sesardelaisla

ikarus23, 2024-02-23:
What button did you use in the software? There is a "Write UID card" and a "Write CUID card". If the "Write UID card" button did work, then we have a plausible answer: The card is a Gen1a card. This type card can only be written with extra tools, not Android. Only Gen2/CUID cards work.

Furthermore, many years ago somebody reported the Samsung J5 as not compatible with MIFARE Classic (see https://github.com/ikarus23/MifareClassicTool/blob/master/INCOMPATIBLE_DEVICES.md). But if you were able to read the tag, than this information might be outdated...

sesardelaisla, 2024-02-25:
> What button did you use in the software? There is a "Write UID card" and a "Write CUID card". 

I tried all options possible. None of them worked for writing.

> Furthermore, many years ago somebody reported the Samsung J5 as not compatible with MIFARE Classic

I guess that’s actually still the problem.