ikarus23/MifareClassicTool #472: Samsung Galaxy S3 (i9300) is really supported at all?

ilgrank, 2024-03-02:
Hi
I've bought for cheap an old Galaxy S3 since it is on the [supported hardware list](https://github.com/ikarus23/MifareClassicTool/blob/master/COMPATIBLE_DEVICES.md)
however, play store wont' let me install MCT since it says it is not supported
I tried manually installing the Release 4.2.2, and it fails with the error "There is a problem parsing the package": usually that means that the APK targets an higher API version than the one of the OS (running the latest official Samsung release: Android 4.3)
Is there anything I can do to run MCT on my Samsung Galaxy S3 besides installing a custom rom like lineage (which, in turn, usually doesn't guarantee NFC functionality?

ikarus23, 2024-03-03:
Hi. With version 1.6.0 of MCT the minimal Android version required was set to 4.4. So Android 4.3 is to old. You can can of course use version 1.5.3 and below, but otherwise Lineage or something similar is required. I would recommend to update your Android because version 1.5.3 is from October 2013... That's over 10 years ago...

You can find all versions of MCT here: https://www.icaria.de/mct/releases/

closed 2024-03-03 by ikarus23

ikarus23, 2024-03-03:
Oh and with regard to your question in the title:
The referenced [COMPATIBLE_DEVICES.md](https://github.com/ikarus23/MifareClassicTool/blob/master/COMPATIBLE_DEVICES.md) file mostly just confirms, that the device supports reading/writing MIFARE Classic tags. So driver, firmware and the NFC chip are compatible. Maybe I should update the wording to make it more clear.

But yes, MCT did run on a Galaxy S3 until version 1.6.0.