ikarus23/MifareClassicTool #484: Feature request: convert partial dump to a different marker or a zero-filled full dump.

vladko312, 2024-05-16:
Some cards have unknown keys for some sectors, while others have Access Conditions like `0870FF` preventing access to some blocks. In this case, MCT marks these blocks with dashes.
It would be a good idea to add a way to quickly replace them wis question marks (flipper zero marks them in this way) or zeroes (as proxmark3 does).
It would be useful for comparing dumps across platforms, as well as for exporting partial dumps in binary format.

ikarus23, 2024-05-16:
Interesting Idea. I don't want to make them `00` bytes because they are not distinguishable from a successful read where the data/keys were zero bytes. But I could live with `??`. Or make it configurable.

All in all it will take some dedication. Unfortunately there are a lot of places where the `--` markers are used. I think even some Regex. A lot of old code by a guy how does not really know how to code ;)

vladko312, 2024-05-17:
> I don't want to make them `00` bytes because they are not distinguishable from a successful read where the data/keys were zero bytes. 

They should not be `00` by default, but it would be useful to be able to quickly convert them into zeroes. Some formats (like binary) and tools require full dumps, as they have no support for those partial markers. Also, it would make it easier to compare dumps to proxmark.

So, it would be a good idea to add a way to convert partial dumps to full ones, even if some data would be lost in the process.

Also, I noticed that for sectors without known keys the editor is not displaying any values, replacing input with the error message. I think, there should still be a way to edit those sectors.