ikarus23/MifareClassicTool #490: Improve reading speed when the keys are known

schdie, 2024-06-29:
Is it possible to make a keymap file with the specific key pair needed for each of the sectors?
In my phone it takes several seconds to read a card when using 32 different keys.

ikarus23, 2024-07-12:
When you have read the card for the first time, you can export the correct keys into a new key file. There is an option for that in the dump editor. This speeds up things significantly.

schdie, 2024-07-12:
The key file I use only has the keys needed for the card I want to read, it has 32 keys in it. It takes about ~20 seconds to read the card. If I save the keys from a dump I get a new key file with the same content and it takes the same amount of time to read the card obviously.

What I'm asking I guess is, if specific keys can be used for specific sectors a/b and if that could possibly speed up the reading.

ikarus23, 2024-07-31:
Oh, you already had only the 32 valid keys in a file. Hmm, I noticed that some devices take way longer than others, but 20 seconds seams really long.

Anyways, unfortunately there is no way to improve read speed. However, you are not the first one with this request. There is a 10 years old (!) issue about that: https://github.com/ikarus23/MifareClassicTool/issues/43. But until now I never had the time or energy to implement it. Sorry. :/

I will close this for now because it's a duplicate of #43.

closed 2024-07-31 by ikarus23