ikarus23/MifareClassicTool #496: One tag was recorded easily, but there is difficulties with the second

TaPo4eK228, 2024-09-04:
I understand that most likely the problem is that I did not figure something out, but I still ask for help, since there is no one else to ask.

The tag of my friend that has already been copied is shown below, I managed to do this simply by using a full copy with a block 0 entry
[image]

And now my tag, which I think looks different in block 0 (or am I stupid anyway).
[image]

personally, it seems strange to me that the uid in the header differs from the first values in the blog 0. I looked and all the tags of my friends that I checked look like in 1 photo

and when I try to make a full copy of my tag I get an error: 
[image]
Unfortunately, my understanding of what to do ends there. but the only thing I managed to do was copy the uid, but it doesn't look like it should: 
[image]

I am often online and if any additional information is needed, I can provide it. Thank you in advance for your help

ikarus23, 2024-09-05:
HI. Yes, this block 0 with the `340005...` looks definitively wrong. Does this happen if you just do a simple read of the tag? What Android device are you using and what version of Android?

Let's try another way. Could you install the NXP TagInfo app and do a full scan on your tag? If you scroll down you should be able to see the read out data from the tag. Has it the same issue like this app?

Regarding your approach of just changing the UID: The 5th byte after the 4 UID bytes is the BCC value. It has to be correct as well. There is a "Bcc Calculator" tool in the "Tools" section of this app. However, if you just want to clone the UID, there is an even simpler way. Goto "Tools -> Clone UID" and follow the instructions.

TaPo4eK228, 2024-09-05:
> Let's try another way. Could you install the NXP TagInfo app and do a full scan on your tag? If you scroll down you should be able to see the read out data from the tag. Has it the same issue like this app?

[image]
[image]
[image]
[0F-76-08-CF_05-сент.-24 21-44-25_taginfo_scan.txt](https://github.com/user-attachments/files/16898936/0F-76-08-CF_05-.-24.21-44-25_taginfo_scan.txt)

My phone is Xiaomi redmi 9t, Android 12, without root

ikarus23, 2024-09-05:
Ok, now I see. This is one of the cards with a 4 byte UID during the discovery (anti collision phase) and a 7 byte UID in block 0. I can't remember from the top of my head what makes them work this way. Are you even sure this is magic gen2 (CUID) tag you are trying to write? You can't just write block 0 on any tag with an Android phone.

TaPo4eK228, 2024-09-05:
> Are you even sure this is magic gen2 (CUID) tag you are trying to write? You can't just write block 0 on any tag with an Android phone.

Of course I'm sure. in the first message in the first picture, this is the recorded gen2 tag. I attach a link where I bought it: https://aliexpress.ru/item/1005004286997129.html?spm=a2g2w.orderdetail.0.0.7aa14aa6PcC2Mq&sku_id=12000028635044657

If you have any ideas, please write. You may find it useful that I have an RC522 board for arduino, but I'm very bad at programming them.

ikarus23, 2024-09-05:
Ok, yes, found the information about these cards again (its section 10.1.1 in the MIFARE Classic EV1 datasheet). These cards can be configured to show different UIDs at different stages. I'm not sure if there are magic tags (gen2, CUID, block 0 writable) that can emulate this behavior. I only know of gen2 tags that can do 4 byte UID or 7 byte UID. Never both.

So unless somebody out there knows of a magic tag that can emulate this exact behavior, I don't think there is anything you or I can do. However, maybe you can trick the system by buying a 7byte gen2 card and only simulate the second 7 byte UID.

ikarus23, 2024-09-05:
The "MFC EV1 CL2 Perso config" feature of USCUID cards looks promising. https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/magic_cards_notes.md#mifare-classic-uscuid
>When configured, the tag behaves like a real Mifare Classic EV1 7B UID tag, and reads UID from backdoor blocks. Otherwise, the tag acts like a 4 byte tag.

This sounds to me as there are solutions to emulate this behavior. However, I don't have a USCUID tag lying around and I'm not sure if the Android NFC API allows me to support them from this app. However, you should be able to configure one with your RC522.

ikarus23, 2024-09-06:
A "ready to use" solution for USCUID cards is the Proxmark3. You can find a cheap version (Proxmark3 Easy)  on various sites. If you flash [this firmware](https://github.com/RfidResearchGroup/proxmark3) you can configure the USCUID card. However, there is still some things to learn.

ikarus23, 2024-09-22:
I will close this for now. Feel free to add more information if you try the USCUID cards. As for now, there is nothing this MCT app do about that configuration.

closed 2024-09-22 by ikarus23