ikarus23/MifareClassicTool #499: Clone Fuel card

pergolafabio, 2024-09-17:
Hi, i found your tooln, have a quick question...

 I want to clone a fuel card, i use that mifare card to charge my car... In order to clone it, i need magic tags with changable UID
But do i also need to buy a reader like an acr122u ? or can it be done also with the NFC reader on the Android itself?

Thnx for your answer

ikarus23, 2024-09-22:
Hi. You can do everything with your phone and this app. No additional reader needed. You just have to make sure you get an "gen2" or "CUID" magic tag (e.g. gen1a tags will not work on Android). These tags are often marketed as "MCT compatible" or "Android compatible" on marketplaces like Aliexpres.

closed 2024-09-22 by ikarus23

pergolafabio, 2024-09-22:
Perfect, thnx for the update, have ordered the gen2 tags indeed

pergolafabio, 2024-09-25:
ok, my smart tag is arrived, and i can clone UID ! thx for that, however, i want to put the sticker behind my pixel 7 phone, so i can use it to swipe at the fuel station
But as soon as its at the back on my phone, far awar from the internal NFC reader, the tag is not recognized by any usb reader...

I tested it, you can see in video below... as soon as my pixel 7 is nearby, the tag is blocked by something, if i move it away, it works .... any idea what can cause this? is this an issue when there is a wireless charging on the phone?

https://photos.google.com/share/AF1QipPhXYMfHHNQh3qyb3SEndCuNsKsVY6RKNSQz2oGeOIDv1zdMYC7o4ALjHcIrplVLg?key=YlpVZ2dDcTVveWc0WDR0eTl3bGJfYkt6LTVvaWd3

ikarus23, 2024-09-25:
Hi. Unfortunately I think this is somewhat to be expected. I can think of two reasons.

1) You phone acts as a reader and is constantly looking for NFC tags. If you hold it close to another reader, said reader might not be able to detect the tag with the interference by a second reader. Maybe do a test with NFC disabled on the phone.

2) You phone is configured with Google Pay. This makes the phone act like an NFC tag. The other reader has now trouble to decide to which tag it should talk to. Having two or more tags in one reader filed causes trouble for a lot of readers and most of them will not read any of the tag (because of collisions).

pergolafabio, 2024-09-25:
Hi, don't think it's possibility 1, since the NFC tag was at the bottom, if move it up , the NFC of my phone was indeed constantly reading it...

I have indeed Google wallet configured, but even if I turn off NFC on my phone, it didn't work :-(