ikarus23/MifareClassicTool #510: Someone please explain the possible causes of different readers behaviors

devtud, 2024-11-24:
I have a tag used to access different areas and buildings in the residence I live in (main residence's gate, few buildings main doors, some interior doors inside the buildings, garbage room' door, the elevators also require the access tag to descend to underground parking).

I want to clone my tag, which is a mifir classic, so I used the app to read it and I successfully generated the dump (with keys and so on).

I bought some writeable (including sector 0 block 0) cheap tags from China and wrote the dump on few of them. I've also read them after, and compared the clone with the original and they're 100% identical on all sectors and blocks. 

The thing I don't understand is that for some reason the clones work only on 2 or 3 doors. The readers on the doors have a led which lights up green of the access is granted, or red if the access is not granted). The thing is that the majority of readers don't even detect the cloned tags as they don't light up the LED green or red. The readers which can read the clones allow access (green), the rest don't emit any led feedback. 

Could someone please explain why this behavior? If it's working on some readers, why it's not working with all the readers? I am assuming all the door readers are the same...

devtud, 2024-11-27:
Here are 4 of the readers. IIRC first 2 work well with the clones, and the others don't. They all look the same to me:
https://ibb.co/WxdVG3q
https://ibb.co/585dbLv
https://ibb.co/NWxdSjq
https://ibb.co/XJWdg65

Here is the original tag: 
https://ibb.co/RyNdtYz

These are 2 clones: 
https://ibb.co/wQ8YZ6x

ikarus23, 2024-11-28:
Oh, it's a vigik reader. These are pretty well known to some people (I personally have no experience with them). My guess is that that it is an antenna issue. Also, these readers were causing trouble in the past for some NFC emulators (e.g. Flipper Zero). But i think it is more likely that the magic/clone tags have bad antennas. You can try to play around with the placement of the tag on the reader. Maybe it is just a weak conneciton.

You can also try to ask at the RFID Hacking Discord Server: https://discord.com/invite/QfPvGFRQxH There is a good chance that someone over there has more experience with vigik readers.

devtud, 2024-11-29:
Thank you for sharing your experience here. I tend to believe it's something related to the readers rather than the tags though. Until now I tried 4 or 5 different looking tags and am of them worked on the same readers and didn't work on the others.

Ok, there is also a chance that the antennas inside the tags are identical and only the shells differ... no idea. 

I will seek for answers on the discord channel then. Thanks again!

closed 2024-11-30 by ikarus23