ikarus23/MifareClassicTool #519: Can't clone UUID

DarthWeber, 2025-01-06:
Hi,

i bought those MCT:
"GEN2 CUID Android App MCT Modify UID Changeable Tag Keyfob NFC 1k S50 13,56MHz Kartenblock 0 Beschreibbar HF 14443A"

and expected, that i can write the uid to it - but it failed when using the uid clone tool:
"Error: Key not correct"

Are those Tags wrong? It sound that they should work (Gen2 Block 0 writeable)

Thanks,
Tom

ikarus23, 2025-01-07:
Hi Tom. Sounds like you got the correct card. At least the description is correct (sometimes on stores like AliExpress, the product description doesn't match the product).

I have some theories about this:
* You might have written to sector 0 (not block 0) before and with that changed the default keys. Can you provide block 3 of sector 0 of the CUID/Gen2 card?
* Are the lengths of the UIDs compatible? You can not write a 7 byte long UID to a 4 byte long CUID/Gen2 card.
* Is the Pixel 8 running the latest Android version? The Pixel 8 had issues with MIFARE Classic before Android 15.

DarthWeber, 2025-01-29:
Thanks for comin back. Unfortunately no. I am not sure anymore, if the tags block 0 is *really* writeable since other users also have this issue.

ikarus23, 2025-01-30:
Hi. Yes, reviews not look good on these ones.

These look way better: https://de.aliexpress.com/item/1005008248133354.html
Also, PISwords is a more trusted source for magic tags. But also a bit more expensive: https://de.aliexpress.com/item/32791886990.html

ikarus23, 2025-01-31:
I will close this for now. Feel free to reopen it, if you still have issues with the new tags.

closed 2025-01-31 by ikarus23