ikarus23/MifareClassicTool #520: 0.05% Difference between dumps?

TheZwen, 2025-04-10:
[image]

I'm only able to make a 99.95% copy of my original mifare classic card. I'm using an Magic Card 2 for the copy. Anyone knows the problem?

ikarus23, 2025-04-11:
Hi. This is called SAK swapping. It's believed to be an anti copy/cloning feature. Have a look at https://gist.github.com/equipter/3022aea4e371e585ff6e46de637e7769. I think you can not get a 100% clone with a gen2 card.

ikarus23, 2025-04-11:
Also: have you tried the copy? There is a chance it might work, if the reader does not check the byte.

TheZwen, 2025-04-12:
> Also: have you tried the copy? There is a chance it might work, if the reader does not check the byte.

ikarus23, 2025-04-13:
Great to hear. I will close this then.

closed 2025-04-13 by ikarus23