ikarus23/MifareClassicTool #526: MCT App does not detect installed External NFC App

kamikazechaser, 2025-08-11:
I have the external NFC app installed (ver 1.04b). When I connect the external PN532 reader via, external NFC detects it. However opening the MCT app, it forces me to go offline, install the external NFC app (no longer available on PlayStore) or exit making it unusable.

ikarus23, 2025-08-19:
Yeah, unfortunately I expected this to become an issue sooner or later. If I can recall from the top of my head, Android does not longer allow apps to search for other apps. But maybe I'm remembering it wrong. I have to have a closer look.

What device and Android version are you using?

kamikazechaser, 2025-08-20:
Redmi note 11 5g, Android 13. Btw I had a conversation with the author of Mtools, he is using https://github.com/mik3y/usb-serial-for-android to talk to various types of readers. His implementation is closed source though.

ikarus23, 2025-08-20:
Ok, it seams like since Android 11, apps have to specify other apps to interact with in order to discover them
* https://stackoverflow.com/questions/6758841/how-can-i-find-if-a-particular-package-exists-on-my-android-device#comment131011431_6758962
* https://developer.android.com/guide/topics/manifest/queries-element?hl=en

I'm not sure if just declaring the package is enough. Could you try this [test version](http://tests.icaria.de/MifareClassicTool-4.2.3-test1.apk)?

ikarus23, 2025-08-21:
Well, are you sure the tags are MIFARE Classic? Do you have another device with NFC? [NXP TagInfo](https://play.google.com/store/apps/details?id=com.nxp.taginfolite) is good at detecting different NFC technologies.

kamikazechaser, 2025-08-22:
Fair enough. Though most lower - mid-range phones don't have NFC especially in Asia/Africa.

ikarus23, 2025-08-22:
> Fair enough. Though most lower - mid-range phones don't have NFC especially in Asia/Africa.

You are right. But at that point it might be better to get a cheap USB NFC reader or a Proxmark3 Easy. The PM3 even has support on Android though Termux I think. 

> Well, are you sure the tags are MIFARE Classic? Do you have another device with NFC? [NXP TagInfo](https://play.google.com/store/apps/details?id=com.nxp.taginfolite) is good at detecting different NFC technologies.

Feel free to reach out if you know for sure, that you tag is MIFARE Classic. In that case External NFC should correctly create an Intent that can be processed by MCT. However, External NFC is not really maintained.