ikarus23/MifareClassicTool #527: Wrong memory size detection for MF Plus EV1

mxcdoam, 2025-08-13:
I was trying to read a MIFARE Plus EV1 2K SL1 card, specifically - from 21st to 26th sectors and got this error: "Error in mapping range: sector out of range". After resetting the mapping range I've got only first 16 sectors read. Seems like the app can't figure out the tag's size.

To reproduce:
1) Put the MF PLUS EV1 2K tag on the antenna
2) Select the appropriate key file
3) Change the value under "Create map for sectors:" to  "from 21 to 26"
4) Press " START MAPPING AND READ  TAG"
5) A pop up will appear with the aforementioned error text.

ikarus23, 2025-08-19:
Yes, seams like the size is detected wrong. Not sure why though. Have you tried the NXP TagInfo app? What size is that app detecting? And what are the ATQA and SAK values (should show up somewhere under details).

Also, you can force MCT to use a specific tag size. Have a look at the settings (three dots at the main menu).

mxcdoam, 2025-08-21:
Yes, I figured three dots menu😊 Forced MCT to expect 40 sectors all the time.
SAK = 0044
ATQA = 08
UIDs are 7 bytes, CL2.
Strangely enough, Flipper Zero detects them as MFC 1k as well. 
Seems like right now Only NXP TagInfo is able to detect the memory size correctly

ikarus23, 2025-08-24:
I opened up an issue. I don't think it will change anything. But it's a least out there. https://issuetracker.google.com/issues/440892514